prusa3d/PrusaSlicer #14042: Use of 3d models as a "mask" for effects

Andful, 2025-01-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Most, if not all surface modifiers(e.g. fuzzy skin) are applied by a paint function. It is non-reproducible and tedious process.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Use a secondary model as a "mask" used to specify the regions where the effect is applied.
**Describe how it would work**
Be as descriptive as your skill level allows you.
Add a function of selecting a second model instead of the paint function. To determine the region where the effect will be applied.
**Describe alternatives you've considered**
None

**Additional context**
Add any other context or paint drawing about the feature request here.

Snuff1eupagus, 2025-01-25:
This is already available and easy to do in Slicer, on several different levels. you have had the ability to use volumes a.K.a (Models) since 2.4 versions.  The (paint tool) for fuzzy skins has only been in Slicer since the 2.9 releases.
Look into and educate yourself modifiers, make sure you have Slicer set to expert.

Andful, 2025-01-25:
Ah thank you. Was not aware of modifier meshes.

closed 2025-01-25 by Andful