prusa3d/PrusaSlicer #14043: unit test fff_print_tests fails

jalapenopuzzle, 2025-01-26:
### Description of the bug

When I run the unit tests, fff_print_tests fails when run in Debug build.

### Project file & How to reproduce

At commit https://github.com/prusa3d/PrusaSlicer/commit/5c7888b11453eb96a95b49dfd4cbd3b56db64eb8

```bash
user@prusa:/home/user/PrusaSlicer/build-debug$ cmake .. -DSLIC3R_STATIC=1 -DSLIC3R_GTK=3 -DSLIC3R_PCH=OFF -DCMAKE_PREFIX_PATH=$(pwd)/../deps/build/destdir/usr/local -DCMAKE_BUILD_TYPE=Debug
...
user@prusa:/home/user/PrusaSlicer/build-debug$ make test
Running tests...
Test project /home/user/PrusaSlicer/build-debug
    Start 1: arrange_tests
1/6 Test #1: arrange_tests ....................   Passed   34.79 sec
    Start 2: thumbnails_tests
2/6 Test #2: thumbnails_tests .................   Passed    0.39 sec
    Start 3: libslic3r_tests
3/6 Test #3: libslic3r_tests ..................Subprocess aborted***Exception: 100.95 sec
    Start 4: fff_print_tests
4/6 Test #4: fff_print_tests ..................Subprocess aborted***Exception: 376.86 sec
    Start 5: sla_print_tests
5/6 Test #5: sla_print_tests ..................   Passed  3161.06 sec
    Start 6: slic3rutils_tests
6/6 Test #6: slic3rutils_tests ................   Passed   16.77 sec

67% tests passed, 2 tests failed out of 6

Total Test time (real) = 3690.87 sec

The following tests FAILED:
	  3 - libslic3r_tests (Subprocess aborted)
	  4 - fff_print_tests (Subprocess aborted)
Errors while running CTest
Output from these tests are in: /home/user/PrusaSlicer/build-debug/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
make: *** [Makefile:71: test] Error 8
user@prusa:/home/user/PrusaSlicer/build-debug$ 
```
Tested on Ubuntu 22.04.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0-linux

### Operating system

Ubuntu 22.04

### Printer model

Snapmaker 2.0

jalapenopuzzle, 2025-01-26:
I have addressed the errors in fff_print_test in PR https://github.com/prusa3d/PrusaSlicer/pull/13886 .

jalapenopuzzle, 2025-02-02:
I have pushed this change again to address comments at https://github.com/prusa3d/PrusaSlicer/pull/14060#issuecomment-2629575140 .

SachCZ, 2025-02-03:
I have reviewed this and it is ready for merge.

closed 2026-09-29 by prusaslicer-bot[bot]