prusa3d/PrusaSlicer #14045: `CHECK( feedrate == Approx(config.opt_float("travel_speed") * 60) )` build test failure on aarch64 and s390x

adelton, 2025-01-26:
### Description of the bug

We started to see
```
4: /builddir/build/BUILD/prusa-slicer-2.9.0-build/PrusaSlicer-version_2.9.0/tests/fff_print/test_retraction.cpp:92: FAILED:
4:   CHECK( feedrate == Approx(config.opt_float("travel_speed") * 60) )
4: with expansion:
4:   7799.0 == Approx( 7800.0 )
4: with message:
4:   move Z at travel speed
```
on Fedora rawhide, on aarch64 and s390x architectures (but not x86_64 and ppc64le), reported as https://bugzilla.redhat.com/show_bug.cgi?id=2341110.

This is with prusa-slicer 2.9.0. The 2.9.0 build has passed (including the tests) on 2025-01-07 (and a couple of time before that) and is in Fedora rawhide, this was seen on a periodic rebuild (to check regressions with new dependencies). Going through the list of packages that are different between the passing build back on 2025-01-07 and now, nothing sticks out -- catch2 is still 2.13.10, for example, and obviously the 2.9.0 source is exactly the same

Attempting to understand the code, it is not exactly clear where the `.f()` (which gets into `feedrate`) comes from, and if relaxing this approximate comparison by one is possibly justified, or if the value should always be much much closer to 7800 and there is some significant problem somewhere.

### Project file & How to reproduce

No project file as this is a build failure.

Steps to reproduce is to build the .src.rpm file from https://koji.fedoraproject.org/koji/buildinfo?buildID=2610767 on latest Fedora rawhide.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 GA + a couple of Fedora build patches at https://src.fedoraproject.org/rpms/prusa-slicer/tree/rawhide

### Operating system

Fedora rawhide

### Printer model

N/A

cryptomilk, 2025-01-27:
Can you try https://github.com/prusa3d/PrusaSlicer/pull/13886 ?

adelton, 2025-04-14:
Thanks for the pointer.

That PR was merged as version_2.9.0-49-g944844131, so rebase to 2.9.1 should have the fix.

adelton, 2025-04-14:
If the assessment that this issue was addressed by https://github.com/prusa3d/PrusaSlicer/pull/13886 is correct, then sure.

closed 2025-04-14 by SachCZ