prusa3d/PrusaSlicer #14053: Prusa Pen Drive broken

ellepeig, 2025-01-28:
Hi there. The Prusa pendrive you sent me with the printer is now broken. Today was impossible to print due to several connection problems. All the files inside are now unreadable. I solved the issue inserting another pendrive in the prusa usb connector, and everithing returmed to work perfectly for now (except for the Prusa pendrive). I've bought the printer 2 weeks ago. Please I ask to resend me another pendrive with the original files inside, conformly to the warrenty. Thanks

theWituch, 2025-01-28:
Wooooow xD

neophyl, 2025-01-28:
You need to contact Prusa support.  This is for reporting software issues.

rtyr, 2025-01-29:
Please contact our tech support through live chat at www.prusa3d.com or by e-mail.

closed 2025-01-29 by rtyr