prusa3d/PrusaSlicer #14054: Some files don't show up.

BeeManCoder, 2025-01-28:
When I go to import 3mf's, they don't show up in the file explorer.

fe60, 2025-01-29:
@BeeManCoder I'm afraid developers won't be able to analyze your issue without further information.

Which version of PrusaSlicer do you use?
Which operating system version PrusaSlicer is running on?
What exactly did you try to do (import or open, menu or short cut, which menu entry)?
Screenshots can provide more details (1. dialog you miss some files in, 2. file explorer showing the files you expect to appear in the dialog).

SachCZ, 2025-01-29:
Hello @BeeManCoder, If you are on Linux I suspect this may be a flatpak permissions issue. Please, see https://github.com/prusa3d/PrusaSlicer/issues/13820#issuecomment-2563593643

SachCZ, 2025-02-27:
No response, closing.