prusa3d/PrusaSlicer #14057: Is there anywhere I can get a compiled x86 linux binary to run this?

eastoncrafter, 2025-01-30:
### Description of the bug

With the new updates shifting to flatpak, there is no where to get a simple static binary anymore, that I can see. I attempted to compile it, but was met with some sort of path issue that's beyond my knowledge at this point. My goal is to use the Prusa Slicer commandline in a project I am making, and I need a simple executable to run from python code. In the end I will be making this a docker container as well, which is why a static build was the intention. Any advice/help?

### Project file & How to reproduce

Not related

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Debian GNU/Linux 12 (bookworm) x86_64

### Printer model

Creality Ender 3

Samueru-sama, 2025-02-01:
Hi you can use [this AppImage](https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous) from the [relevant issue](https://github.com/prusa3d/PrusaSlicer/issues/13653).

I noticed the CI in probonos fork started to fail and now only an old release is available 🤔

BrixSat, 2025-02-08:
I miss appimage files :) that is the only thing i use.

CRCinAU, 2025-02-18:
Prusa have shown that they just don't care anymore. They'd rather push their integrations and move more towards a fully captive environment than create a quality product.

eastoncrafter, 2026-09-18:
As the builds for v3 haven't started rolling out yet, I can't confirm that this issue will be solved