prusa3d/PrusaSlicer #14058: sync visually M601 (break) with the visual slicing preview

ufgum, 2025-01-30:
### Description of the bug

If you visually scroll thru the layers to make a break, to insert a nut you will insert the break in the last layer which is seen open, that's for is visuality, but prusa slicer or firmware makes the break before layer. That means I need to go to the first closed layer and insert a break.

- If I insert a break, there is no popup, telling me: "Be careful, the break is on beginning of the layer, do not doing it visually"

But in my Opinion therefore is visuality.

### Project file & How to reproduce

make a poket, slice it and insert visuality a break. It will be one layer to early and ma distroy the print.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

arch linux

### Printer model

prusa XL

Gonsior85, 2025-02-11:
I've noticed this too and I know why it happens. I think we should combine our issues into one. https://github.com/prusa3d/PrusaSlicer/issues/14080

ufgum, 2026-09-02:
I will check if here something changes...