prusa3d/PrusaSlicer #14061: Use system browser for Prusa Connect login

abatkin, 2025-01-30:
PrusaSlicer should give the option of using the system browser for login to PrusaConnect.

**Benefits of this approach:**
 - Users often have password managers attached to their browsers, making the login flow easier, faster and more secure. Using the current embedded dialog, a user would need to launch a password manager and copy/paste the password between applications, which is cumbersome, time consuming, and insecure (since the password must traverse the system clipboard)
 - Browser may already have an active/secure session with the Prusa login system, further reducing friction
 - In general, it's not a good security practice for users to be in the habit of entering credentials directly into individual applications when a more secure alternative is available. Web browsers have many security features and capabilities that have evolved over the years, so offloading this security sensitive task to a tool (web browser) that can handle this security-sensitive operation is a great idea

In the case where a logged-in browser session is already available, one could imagine a case where the login flow would literally be as simple as:
1. Click "Login" button in PrusaSlicer
2. Browser launches, presents a button to the user to "Allow"
3. Browser responds "You may now close this window" (and PrusaSlicer is logged in)

Even if it's a brand new browser, step 2 above becomes "login form" and it only adds one extra step (and can take advantage of a password manager). Whereas with the current flow, anyone using a password manager has a number of additional steps (launch password manager, look up appropriate secret, copy to clipboard, switch back to PrusaSlicer and paste password, repeat for MFA token)

dwbmb, 2025-02-02:
I am for this idea. It used to be that way by the way! PrusaConnect loges me out multiple times a day at it is a huge pain to login with secure password in-app without password manager like in browser. Please, if you do not consider fixing logout issue, take login via browser back!

nicopasla, 2025-04-21:
I'm also for this idea, its a pain to copy/paste passwords every time it logout for no reasons

closed 2026-09-29 by prusaslicer-bot[bot]