prusa3d/PrusaSlicer #14063: Flatpak version of prusaslicer 2.9.0 exits when switching to physical printer tab

Petersat7funde, 2025-01-30:
### Description of the bug

Flatpak version of prusaslicer 2.9.0 exits when switching to physical printer tab
It switches to the printer web screen, which is stalled and quits.

And simply starting it, brings up errors. 

Its the only flatpack install in the system.

So its not usable at all :((

Need to go back to the latest appimage to be able to print...

### Project file & How to reproduce

 flatpak run com.prusa3d.PrusaSlicer 
--------------------------------------------------------------------------
Message: 17:35:42: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
peter@White:~/Downloads> Gtk-Message: 17:35:42.324: Failed to load module "canberra-gtk-module"
Gtk-Message: 17:35:42.347: Failed to load module "canberra-gtk-module"
Gtk-Message: 17:35:42.348: Failed to load module "canberra-gtk-module"
[2025-01-30 17:35:43.494318] [0x00007f94d249e740] [error]   UserAccount: Failed to read token - no datafile found.
05:35:45 PM: Debug: window wxTreeCtrl@0x563016c08b50 ("treeCtrl") lost focus even though it didn't have it
05:35:45 PM: Debug: window wxTreeCtrl@0x563016c08b50 ("treeCtrl") lost focus even though it didn't have it
05:35:45 PM: Debug: window wxTreeCtrl@0x563016c08b50 ("treeCtrl") lost focus even though it didn't have it
05:35:45 PM: Debug: window wxTreeCtrl@0x563016c08b50 ("treeCtrl") lost focus even though it didn't have it
05:35:45 PM: Debug: window wxTreeCtrl@0x563016c08b50 ("treeCtrl") lost focus even though it didn't have it
05:35:45 PM: Debug: window wxTreeCtrl@0x563016c08b50 ("treeCtrl") lost focus even though it didn't have it
Gtk-Message: 17:35:54.245: Failed to load module "canberra-gtk-module"
Gtk-Message: 17:35:54.271: Failed to load module "canberra-gtk-module"
Gtk-Message: 17:35:54.272: Failed to load module "canberra-gtk-module"
[2025-01-30 17:37:02.220641] [0x00007f94493f96c0] [error]   Reversing even wall line causes it to be printed CCW instead of CW!
Gtk-Message: 17:37:53.043: Failed to load module "canberra-gtk-module"
Gtk-Message: 17:37:53.067: Failed to load module "canberra-gtk-module"
Gtk-Message: 17:37:53.068: Failed to load module "canberra-gtk-module"

flatpak list
Name                                             Anwendungskennung                              Version            Zweig                Installation
PrusaSlicer                                      com.prusa3d.PrusaSlicer                        2.9.0              stable               system
Mesa                                             org.freedesktop.Platform.GL.default            24.3.1             24.08                system
Mesa (Extra)                                     org.freedesktop.Platform.GL.default            24.3.1             24.08extra           system
openh264                                         org.freedesktop.Platform.openh264              2.4.1              2.4.1                system
GNOME Application Platform version 47            org.gnome.Platform                                                47                   system

System is

cat /etc/os-release 
NAME="openSUSE Tumbleweed"
# VERSION="20250108"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20250108"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
# CPE 2.3 format, boo#1217921
CPE_NAME="cpe:2.3:o:opensuse:tumbleweed:20250108:*:*:*:*:*:*:*"
#CPE 2.2 format
#CPE_NAME="cpe:/o:opensuse:tumbleweed:20250108"
BUG_REPORT_URL="https://bugzilla.opensuse.org"
SUPPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo-Tumbleweed"

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 flatpak

### Operating system

openSUSE Tumbleweed 20250108

### Printer model

THE100

henkson, 2025-02-02:
Same (or very similar) problem on MX-Linux 23.5

```
$ flatpak run com.prusa3d.PrusaSlicer
--------------------------------------------------------------------------
Message: 22:18:04: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
Gtk-Message: 22:18:04.211: Failed to load module "xapp-gtk3-module"
Gtk-Message: 22:18:04.211: Failed to load module "canberra-gtk-module"
[2025-02-02 22:18:04.452750] [0x00007f481db7f740] [error]   UserAccount: Failed to read token - no datafile found.
22:18:04: Debug: window wxTreeCtrl@0x558f60bf7c90 ("treeCtrl") lost focus even though it didn't have it
22:18:04: Debug: window wxTreeCtrl@0x558f60bf7c90 ("treeCtrl") lost focus even though it didn't have it
22:18:04: Debug: window wxTreeCtrl@0x558f60bf7c90 ("treeCtrl") lost focus even though it didn't have it
22:18:04: Debug: window wxTreeCtrl@0x558f60bf7c90 ("treeCtrl") lost focus even though it didn't have it
22:18:04: Debug: window wxTreeCtrl@0x558f60bf7c90 ("treeCtrl") lost focus even though it didn't have it
22:18:04: Debug: window wxTreeCtrl@0x558f60bf7c90 ("treeCtrl") lost focus even though it didn't have it
Gtk-Message: 22:18:06.808: Failed to load module "xapp-gtk3-module"
Gtk-Message: 22:18:06.808: Failed to load module "canberra-gtk-module"
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
```

adejong1952, 2025-02-11:
I understand it needs fixing, but 2.9.0 works fine as long as you don't click on the Physical Printer Tab. I use a browser to view mainsail and control printer. A bit inconvenient but until the problem is fixed it does work. I don't think the linux version is a high priority for the prusa team.

CRCinAU, 2025-02-18:
bUt FlAtPaK wIlL fIx AlL tHe PrObLeMs!

SachCZ, 2025-03-03:
Hello, it probably is not a Flatpak issue, but a PrusaSlicer issue, in any case we are actively investigating it, sadly it is very hard to reproduce.

Also please note that Linux version **is the same priority** as the versions for the other platforms.
I am daily driving linux for both my personal and work needs, but comparing the three main platforms, distributing our software for Linux is the most complex and takes up most of developer time to maintain (agnostic of the distribution method being used), yet we are still actively trying to keep it up.

adejong1952, 2025-03-03:
@SachCZ Thank you for letting us know this issue is being worked on. Would be interesting to know of any distros where PrusaSlicer functions correctly.

SachCZ, 2025-03-03:
@adejong1952, for your specific issue. Is the problem with Moonraker?
In that case it is possibly a duplicate #13835.

You may be able to side stepping it by running `WEBKIT_DISABLE_COMPOSITING_MODE=0 WEBKIT_DISABLE_DMABUF_RENDERER=0 flatpak run com.prusa3d.PrusaSlicer` as suggested by @AlexanderS in the issue I just linked.

@CRCinAU, If you feel brave I encourage you to to create the integrations we currently have (Physical printer tab - PrusaLink, Moonraker, Octoprint, etc. + Printables, PrusaConnect) using API level calls and create a pull request. I personally see it as **pretty impossible** to achieve in reasonable time and than subsequently **maintain it forever** while any of the APIs can change at any time, but that is just my **opinion**.

I also feel like there might be a slight missunderstanding about issues we had with Appimage, because **this one is not one of them**. As I stated before this issue probably is not Flatpak specific. For detailed explanation why we decided to move to Flatpak see Reasons for moving to Flatpak in the [2.9.0-alpha1 changelog](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). **Webkit runtime problems are orthogonal to the problems of bundling libc and webkit for distribution alongside an application**, but I am sure you (@CRCinAU ) understand that.

adejong1952, 2025-03-03:
@SachCZ Thanks again for the reply. Running this command does open PrusaSlicer correctly and it functions properly, including the Physical Printer tab. No crashes at all. FYI this is the information provided by terminal.
++++++++
andre@valand-mint:~$ WEBKIT_DISABLE_COMPOSITING_MODE=0 WEBKIT_DISABLE_DMABUF_RENDERER=0 flatpak run com.prusa3d.PrusaSlicer
--------------------------------------------------------------------------
Message: 07:52:20: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
Gtk-Message: 07:52:21.013: Failed to load module "xapp-gtk3-module"
andre@valand-mint:~$ Gtk-Message: 07:52:21.137: Failed to load module "xapp-gtk3-module"
[2025-03-04 07:52:23.168859] [0x00007df031992e40] [error]   UserAccount: Failed to read token - no datafile found.
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
07:52:25: Debug: window wxTreeCtrl@0x6010bb00ff40 ("treeCtrl") lost focus even though it didn't have it
[2025-03-04 07:52:27.501925] [0x00007df02d6006c0] [error]   Error getting: `https://preset-repo-api.prusa3d.com/v1/repos/non-prusa-fff/Geeetech/125X130.svg`: HTTP 404, <?xml version='1.0' encoding='UTF-8'?><Error><Code>NoSuchKey</Code><Message>The specified key does not exist.</Message><Details>No such object: prusa3d-content-prod-14e8-preset-repo-api-public/non-prusa-fff/Geeetech/125X130.svg</Details></Error>
[2025-03-04 07:52:30.726120] [0x00007df02d6006c0] [error]   Error getting: `https://preset-repo-api.prusa3d.com/v1/repos/non-prusa-fff/Geeetech/180X150.svg`: HTTP 404, <?xml version='1.0' encoding='UTF-8'?><Error><Code>NoSuchKey</Code><Message>The specified key does not exist.</Message><Details>No such object: prusa3d-content-prod-14e8-preset-repo-api-public/non-prusa-fff/Geeetech/180X150.svg</Details></Error>
+++++++++++
The info about Geeetech is a bit confusing, but not troubling.

SachCZ, 2025-03-04:
Thank you, all this is valid. I am closing this as a duplicate, to continue the conversation in the original issue #13835 .

closed 2025-03-04 by SachCZ