prusa3d/PrusaSlicer #14064: Arachne Perimeter Generation

TpForMyBungholio, 2025-01-30:
### Description of the bug

Slicing this particular model (~280mm in length) with arachne perimeters causes some apparent randomness in portions of the perimeter path. Classic perimeters doesn't appear to have the same issue.

The model shown in the screenshot is proprietary, so I included a slightly different model in the project file. The pattern in question may look slightly different than what's shown in the screenshot due to this change.

[image]

### Project file & How to reproduce

[Arachne Issues.zip](https://github.com/user-attachments/files/18609081/Arachne.Issues.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 10 Pro

### Printer model

Custom Delta

Tupson444, 2025-01-31:
This could be caused by varying wall thickness of the part, because arachne changes extrusion width based on that. The result is inconsistent outer layers.

I had opened a feature request that might be related to this: #13850

TpForMyBungholio, 2025-01-31:
I'd say it's not exactly the same thing as your issue given that this issue persists even with a single perimeter/ when there are no adjacent print surfaces that it's trying to resolve. That, and extrusion width appears to remain constant, even in this updated example.

[image]

neophyl, 2025-01-31:
You should change the preview to display extrusion width rather than feature type to get a proper appreciation of what Arachne  is doing.

caufang, 2025-02-06:
Arachne appears to use some path simplification for tracing its extrusion paths.

In the model from this ticket, the first several millimeters of model have a very shallow curve, which Arachne seems to shortcut by using a single extrusion move. You can see this by moving the horizontal slider and watching the tool move the entire length in one g-code entry.

Further up the model the curve becomes deeper and must pass some threshold because then Arachne attempts to follow the curve of the model. This is where you see the "shelf" at about z = 3 mm, where the next external perimeter is almost a full extrusion width inward from the previous layer, at the x-middle of the model.

As the layers continue up, Arachne appears to again skip some portions that are very shallow curves, leading to the ribs appearing at different places.

I was able to duplicate these issues by creating a test file that has a shallow curve that grows deeper over an arc.

[arachne_test.zip](https://github.com/user-attachments/files/18683975/arachne_test.zip)

Tupson444, 2026-09-19:
maybe still relevant