prusa3d/PrusaSlicer #14065: Print part fine details show in PrusaSlicer are washed out in 3D print

zorro2055, 2025-01-31:
### Description of the bug

Hello, 
This first image below shows what PrusaSlicer predicts what will be printed.  Notice all the small white stars.  The are pits.  Every so maybe layers, the filament is switched to get this. 

The second print shows most of those stars are not visible, essentially printed over when they should not be.  This seems like this could be a bug in the slicer or maybe more likely, the firmware in which case I could refile this bug there if needed. 

I'm printing with a MK4S with an MMU3.

[image]

[image]

Hopefully, this is an easy fix. 

Thanks.

### Project file & How to reproduce

Will update from computer in a minute with this file.

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11

### Printer model

Prusa MK4S with MMU3

zorro2055, 2025-01-31:
Project file attached

[Nebula_Front_v2_more_stars_visible_200x160.zip](https://github.com/user-attachments/files/18615608/Nebula_Front_v2_more_stars_visible_200x160.zip)

u89djt, 2025-01-31:
(From a fellow user) What you've shown indicates that your white filament is not opaque at a layer thickness of 0.16 (2x0.08)
Print a test object of just a base layer of the white filament to see how it transmits and reflects light.

u89djt, 2025-01-31:
Right, got you. So, speculating: use a flash and it'll look the same. You seem to be relying on ambient lighting, which would have to get down that tube whose first half is dark and so will absorb illumination. The slice preview isn't ray traced :)

zorro2055, 2025-01-31:
Yeah, shining my cell phone flashlight on it, the second smaller star can't really be seen partly due to glare caused by the flashlight on the surrounding surface. 

Really, I think what is going on is the actual hole diameter in the to see the white in the part is smaller than what PrusaSlicer calculates.  For both stars actually.  The first star is more visible, but still appears to have a smaller diameter than what is in PrusaSlicer.  It could be a printer firmware issue of course.....  Or tolerances on the printer just can't be held tightly enough....

u89djt, 2025-01-31:
Let's keep going. You've got glare, so we still haven't assessed the effect of the dark tunnel the light has to get down.
I think a correctly sized aperture with dark edges would lead to the phenomena you've seen so far.
There will certainy be some variation in the precise location of plastic forming the aperture, but you'd only be seeing a tenth of a millimetre at worst.
[image]
If you don't have appropriate tools, push some clay into the surface and look at the size of the impression of the hole?
If the hole size is wrong, check the diameter of your filament and set the extrusion multiplier?
You could consider XY compensation of the darker filaments. That can set by heigh I believe.

zorro2055, 2025-01-31:
Ah, ok, you are measuring the smaller hole partway up.  So I just measured that hole with a pair of calipers and it measured about 0.7mm wide which is the width at the bottom.  So I guess that is right.
[image]

So with my camera at maximum zoom using the flashlight, I was able to get a better picture, although not great.  See pic below.
[image]

It actually looks red.  It ultimately looks red to my eye if I angle it enough.  The red layer looks okay in the slicer, but maybe it is off with printing?  I checked the width of my red filament.  It is good at 1.73mm.
[image]

**So what I think I will do is reprint this pausing the print right before color transitions and measure the holes that are being created and taking pics.  Then I can report back.**

For reference, here the larger star measured about 1.37mm.
which is pretty close to the bottom of the pit measuring 1.4mm.
[image]

And here is a pic of the larger star with camera fully zoomed in with flashlight on.
[image]

neophyl, 2025-01-31:
Yeah as we know every hole printed is undersized.  Its why there's been so many calls for a hole shrinkage parameter in PS and why we take this tendency into account when designing stuff like screw holes.  

This is just that raised to the Nth degree I think.

zorro2055, 2025-01-31:
Ah, thanks both for info.  I will do the study I mentioned just to get that documented since that might help how to better define an auto hole shrinkage parameter for the smaller holes.  Then I can figure out where to move this so that info is visible.  Or close it if the info makes no sense.  Since it will take 8-10 hours I think print out those areas, I'll update with that info later today or tomorrow.

zorro2055, 2025-02-01:
Okay, so from pausing just before color changes that happen at certain layers, there is definitely something wrong happening at the red layer where it basically fills up the smaller holes where the other colors do not.  It seems like the slicer preview is correct, but perhaps the gcode export is not?  Or there is something weird with the filament?  Or some sort of printer firmware bug?  Something else?

Let me know if other info is needed.

All the details below (small star first, big star follows for reference):

**Small star (had issue)**
First color is white, i just took a screenshot from PrusaSlicer because nothing interesting happens here.
[image]

Second color is red.  Here is the screenshot from PrusaSlicer showing it is fine since the white is exposed.
[image]
But here is an angled image showing the actual print basically just covers up the hole so the white is essentially not visible, **so it broke somehow**.  It just seems to go right over the white although there seems to be a circular feature there.
[image]
Here is the zoom pic showing the actual print showing the same thing.
[image]

Third color is white again.  These holes were fine, but they only show red since the red color had an error somehow.  Here is the screenshot from PrusaSlicer showing it is fine.
[image]
Here is an angled image showing the actual print is fine (except only red is visible due to messed up red color).
[image]
Caliper roughly showing size of hole which is closed to measured 0.7mm in PrusaSlicer.
[image]
Here is the zoom pic showing the actual print showing the same thing.
[image]

Third color is glittery blue.  These holes were fine, but they only show red since the red color had an error somehow.  Here is the screenshot from PrusaSlicer showing it is fine.
[image]
Here is an angled image showing the actual print is fine (except only red is visible due to messed up red color).
[image]
Here is the zoom pic showing the actual print showing the same thing.
[image]

The only color after this in this area is black.  That is represented by pictures in the earlier posts in this open issue.

**Large star (no issue)**
First color is white, i just took a screenshot from PrusaSlicer because nothing interesting happens here.
[image]

Then is the red color.  Here is the screenshot from PrusaSlicer showing it is fine since the white is exposed.
[image]
Here is an angled image showing the actual print is fine.
[image]
Here is the zoom pic showing the actual print showing the same thing.
[image]

Then is the white color again.  Here is the screenshot from PrusaSlicer showing it is fine since the white is exposed.
[image]
Here is an angled image showing the actual print is fine.
[image]
Caliper roughly showing size of hole which is closed to measured 1.4mm in PrusaSlicer
[image]
Here is the zoom pic showing the actual print showing the same thing.
[image]

Then is the glittery blue color.  Here is the screenshot from PrusaSlicer showing it is fine since the white is exposed.
[image]
Here is an angled image showing the actual print is fine.
[image]
Here is the zoom pic showing the actual print showing the same thing.
[image]

The only color after this in this area is black.  That is represented by pictures in the earlier posts in this open issue.

zorro2055, 2025-02-01:
I just compared the gcode file in the small star red area, and it looks the same as the code shown in the Prusa Slicer GCode viewer, so maybe this is a firmware issue, and I should file this there?  I guess it could be be a filament issue, but it seems too great for that?

[image]
[image]

zorro2055, 2025-02-01:
Well, this is looking less likely to be a PrusaSlicer issue, but I am going to swap out that red with the blue filament at those layers messing up to see if indeed the red filament is somehow causing the issue just for full documentation.

zorro2055, 2025-02-02:
Sigh, well, it does look like something weird is going on with the red filament.  Printing the red layers as the blue filament did not entirely cover up the hole.  See those blue pics below.

Maybe I'll try to use a super slow small perimeter speed to see if the red likes that.

Anyway, I'll close this now.

I was trying to think of a way to make small perimeters more robust against uncooperative filaments, but nothing came to mind.

Small star:
Glittery blue in place of red (0.70 mm is size of hole):
[image]
[image]
[image]

Large star (for reference):
[image]
[image]

closed 2025-02-02 by zorro2055