prusa3d/PrusaSlicer #14066: Directories not seen by version 2.9.0

jehojakim, 2025-01-31:
I just installed PrusaSlicer version 2.9.0, on OpenSuse Tumbleweed.
When I want to open a file, PS only seems to see directories **owned** by the user, not everything **visible** to the user. Which is a problem here, because I keep a lot of documents not in /home/<user>, but in /home/<some directory to share>.
That directory is not seen by PS, so I can't open anything.
Which renders this version pretty useless for me.

Searched in the configuration, but found nothing to change that behavior.

The last 2.8 versions (and before) did not have this problem.

Is this behavior by intention, or a bug? Or is there something I can do to change it?

regards, jehojakim

u89djt, 2025-01-31:
(From a fellow user) could this be relevant? https://github.com/prusa3d/PrusaSlicer/issues/13820#issuecomment-2563593643

jehojakim, 2025-01-31:
Yes, there I got the solution that works for me: issue "flatpak override --user --filesystem=host com.prusa3d.PrusaSlicer" before issuing the flatpak run command. The flatpak seal option needs another flatpak install, and I want to stay as far from flatpak as possible.

Thanks a lot for this comment!

u89djt, 2025-01-31:
Great! A couple of people have run into the issue, so I don't know whether it's best to leave it open or close it.

SachCZ, 2025-02-02:
There is pinned issue about this. Closing.

closed 2025-02-02 by SachCZ