prusa3d/PrusaSlicer #14068: Filament max volumetric speed is exceeded

FlyingRatBull, 2025-02-01:
### Description of the bug

I set the *Max volumetric speed* of my filament to `16.035` mm³/s and sliced/printed a part. My Extruder started to skip and after slowing it down I analyzed the sliced model in Prusa and discovered that it exceeded the set limits by ~3.5% with `16.630` mm³/s.

### Project file & How to reproduce

1. Slice it as normal (without adaptive layer height)
2. Look at layer 56, step 26681 / the part where it prints the front handle
3. Some of the handle parts have a *Volumetric flow rate* of above 16.035

Project file: [Drawer.zip](https://github.com/user-attachments/files/18628938/Drawer.zip)

# Filament settings

**Note that the screenshot shows 0 for Max non-crossing infill speed and Max crossing infill speed which I set to the same value as above afterwards and got the same results**

[image]

# Exceeded flow in sliced model

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Ubuntu 22.04 LTS

### Printer model

Anycubic i3+

u89djt, 2025-02-02:
From a fellow user doing triage: it looks like you posted a different 3mf file?

[image]

FlyingRatBull, 2025-02-02:
> From a fellow user doing triage: under windows, it looks like you posted a different 3mf file?
> 
> [image]

You're right @u89djt, something got mixed up.
I've updated the image to show *layer 56/step 26874*. and also chose to show the *Actual volumetric flow rate* which is still too high.

FabioM3, 2025-03-31:
The problem is sorted out in the orca slicer(but not fixed yet), the problem occurs because the math relies on an fixed extrusion multiplier of 1.0.

If you change it the end result will be out.

closed 2026-09-29 by prusaslicer-bot[bot]