prusa3d/PrusaSlicer #14076: Multi material sequential printing

bojorno1, 2025-02-03:
Currently, PrusaSlicer does not allow multi material sequential printing, as it says something along the lines of 'Wipe tower not supported for sequential printing'

**Describe the solution you'd like**
I would like it to allow color change between objects, so that way it could print one set of objects in one color, and another set in a different color, all in the same print.

**Describe how it would work**
When sequential printing is enabled and two objects with two different colors is detected, it will change colors between the objects by purging somewhere, preferably outside the print area or 
near the initial purge line.

**Describe alternatives you've considered**
The printer could also have a pause print command in between both objects automatically.

**Additional context**
This could be useful, especially on large scale multi material 3D printers, by allowing the user to print multiple different parts in multiple different colors or materials, without user intervention.

u89djt, 2025-02-03:
(From a fellow user) Purging near the initial purge line would knock everything over.
I think (just another user) the right answer right now is to add your own small purge object - a long straight line to get rid of blobs - between each main object as a separate sequential print if you need to purge the tool. Give it a wide extrusion width to make sure it sticks to the bed at some point just like the standard purge.

Alienbliep, 2025-11-16:
I would really love this feature too.
Now i am adding a plate of 20x20x1mm to purge in between. But you have to order the parts correct and arrange them youself.
It is a lot of work but necessary if the printer is unantended.
I know bambu studio can do it automaticly. Even Per buildplate!! So it won't whine on other buildplates that the wipetower is enabled etcetc.

archiebug, 2026-01-09:
I had a this youtube short sent to me that pretty much covers the use case on seq printing with color change between completed objects.

https://youtube.com/shorts/dw8l08P5iNc?si=ATMuGSVyrpfoV_jy

This would improve efficiency when priting the same object that needs to be different colors.

rcmaniac25, 2026-09-13:
Still relevant for when sequential printing is supported.

Alienbliep, 2026-09-22:
Yes i am still waiting on the feature.....