prusa3d/PrusaSlicer #14080: Incorrect display of colors in the layers bar

Gonsior85, 2025-02-03:
Hello, the problem is that the colors of the layers in the cut preview and in the layer bar are not the same. Everything is shifted by 1 layer. Additionally, the first layer (from 0-0.2) is not visible in the layer bar. The layer bar does not start from zero but from 0.2, hence the problem. It is difficult to describe it in words, but it is perfectly visible in the screenshots. It is also best to create an object with dimensions of 20x20x0.8 and cut it into 4 layers, then change the color of each of them and compare the cut preview with the layer bar

[image]

In the photo above one yellow layer is missing. The orange layer is from 0-0.2 and on the layers bar from 0.2-0.4 where there is actually a black layer etc.

[image]

Here, the second black layer is missing from the layers bar.

I use PrusaSlicer 2.9.0 and Win 11 x64

u89djt, 2025-02-03:
You need to zip the 3mf project file to post it.

Gonsior85, 2025-02-04:
[Kształt-Sześcian.zip](https://github.com/user-attachments/files/18648170/Ksztalt-Szescian.zip)

Gonsior85, 2025-03-05:
How long does it take for someone to be interested in my problem?

u89djt, 2025-03-06:
(From a fellow user) I would /guess/ that it's right down at the bottom of the priority list because it doesn't actually stop you doing anything. It's definitely wrong, and they'll want to fix it, but it looks like a classic off-by-one somewhere and fixing those can eat a lot of time. You'll recognize that the issue wasn't closed while others have been.
Remember that the numbers on the top layer and bottom layer display sliders are the height of the nozzle while extruding that layer, and the ordinal number of that layer.
https://github.com/user-attachments/assets/0fcd9623-b938-49aa-b59b-56715fa3ea1f

Gonsior85, 2025-03-08:
the black color in the layer bar is at a height of 0.4 to 0.6 mm and is actually printed at a height of 0.2 to 0.4 mm. In addition, the layer bar only displays three layers and the model actually has four layers.

closed 2025-11-05 by Gonsior85