prusa3d/PrusaSlicer #14082: Prusa Slicer sharp corner artifacts

ajeans2, 2025-02-03:
On this particular object, Prusa Slicer was unable to accurately follow the edge contours resulting in "jaggies." Is there a setting I'm missing?

[image]

[image]

ajeans2, 2025-02-04:
> This is caused by Arachne slicing having to decide between 1 widened or 2
> thinned perimeters. Switch it back to classic perimeters.
> I have this same problem with thin propellers.

Thanks! I'll give it a try.

Albert

u89djt, 2025-02-04:
(From a fellow user)  You need to zip the 3mf file to post it. If it was just the switching between number of perimeters suggestion, there would be no gaps: it would use thin perimeters. Let's look at the project and investigate thoroughly. A lot of folk find things like this confusing, so it might be worth hanging on to this example if it's illuminating.

ajeans2, 2025-02-04:
Thanks, that did the trick!

u89djt, 2025-02-04:
It did **a** trick.

closed 2026-09-29 by prusaslicer-bot[bot]