prusa3d/PrusaSlicer #14083: macOS PSlicer does not store window size state when shutdown

crabbedh, 2025-02-04:
### Description of the bug

Running macOS Sonoma 14.7, but don't think that is issue. PSlicer always starts in a very small window requiring resizing every time. No matter how I resize, save, shut it down, it always starts up in in window that is about 1600x1000, which is tiny on my 4K display. This is a new behaviour since the last 2 iterations. (I've used for 3 years). I believe developer does not save the closed window size to the PLIST file.

### Project file & How to reproduce

Just go to any macOS and open PS, resize.. save to desktop or not.. doesn't matter.  Close by any method. Reopen and is back to 1600x1000. Screenshot of no value as you have no reference. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

macOS 14.7.1 Sonoma

### Printer model

MK3S+

neophyl, 2025-02-04:
I replied to your forum post but check you have the Preferences setting Restore window position on start ticked

[image]

crabbedh, 2025-02-04:
@neophyl.. Thanks so much.. long time Mac user.. but never thought to look there.. That was issue

closed 2025-02-04 by crabbedh