prusa3d/PrusaSlicer #14084: Rear seam position plus scarf joints causes erroneous outer perimeters

stur0067, 2025-02-04:
### Description of the bug

When using rear seam position and scarf seams on contours, I am getting incorrect perimeters in some areas
[image]
It does not happen for aligned seam position.
This effect does not occur when I try my best to match the settings in Orca slicer 2.2.0.

### Project file & How to reproduce

[example.zip](https://github.com/user-attachments/files/18650767/example.zip)
Slice the model and examine perimeters in the 27.5 to 33.5mm height range.
The layers with artifacts look to be having their entire outer perimeter printed at a different z value.

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 10

### Printer model

MK4

SachCZ, 2025-02-04:
Let us know if this is a duplicate of #13905 so we can close this as duplicate. In the meantime I wish you happy printing.

stur0067, 2025-02-05:
It seems likely that this is a duplicate. I had noticed it as incorrect Z values for the outer perimeter but I do see similar lines in the volumetric view

SachCZ, 2025-02-05:
Thank you for clarification.

closed 2025-02-05 by SachCZ