prusa3d/PrusaSlicer #14104: SD card writting issue

Maciejgvd, 2025-02-08:
I found a malfunction in PrusaSlicer when I inserted the SD card to the PC. The writing protection was detected as on(but it's  off/my hardware  defect). I needed to reconnect the SD card, but PrusaSlicer still detected the write protection is on (despite the fact that the socket detects an already unlocked write). 
I had to exportthe g-code to the desktop and send it manually to the SD card.
I'm not sure if this was present from the beginning or if it appeared after some update.
Thanx.

kimtwio, 2025-02-10:
Hi

Please try another SD card. So far, I only got a write protection warning when my SD card was starting to fail. I suspect yours is also at the end of it's lifespan :(

Maciejgvd, 2025-02-11:
I know there is something wrong with my SD card, but PrusaSlicer not working 100% corectly too, which is visible thanks to my broken SD card ;]

Maciejgvd, 2025-02-13:
Sorry for the confusion. I just found out that Prusa Slice also writes to SD card. After reinserting the card. Thanks again

closed 2026-09-29 by prusaslicer-bot[bot]