prusa3d/PrusaSlicer #14105: Typo - All Print profiles for the Prusa Mini ends with "@MINIIS"

gaellafond, 2025-02-08:
### Description of the bug

There is a Typo in the Print profiles for the Prusa Mini. They all ends with "@MINIIS"
- 0.10mm FAST DETAIL @MINIIS 0.4
- 0.15mm SPEED @MINIIS 0.4
- 0.15mm STRUCTURAL @MINIIS 0.4
- 0.20mm SPEED @MINIIS 0.4
- 0.20mm STRUCTURAL @MINIIS 0.4

It should be "@MINIS" or "@MINI", but not "@MINIIS" with a double "I".

[image]

### Project file & How to reproduce

1. Setup PrusaSlicer 2.9.0 with Prusa Mini+ printer.
2. Go to Printer Settings
3. Open the Preset dropdown

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.0 (installed using FlatPak on Linux)

### Operating system

Ubuntu 24.04

### Printer model

Prusa MINI+

u89djt, 2025-02-08:
"IS" stands for "Input Shaper". You can close the issue yourself (from a fellow user).

[image]

closed 2025-02-08 by rtyr

gaellafond, 2025-02-10:
Thanks @u89djt for looking into this.
Sorry for the confusion.