prusa3d/PrusaSlicer #14111: Bed number removes part of the model layer preview

unkn-one, 2025-02-10:
### Description of the bug

When using multiple beds, bed number cuts part of the model preview in the first 3 layers, but only when looked from top.
[image]
[image]

I would expect number to be properly underneath the model like in the >4 layer preview.
[image]

### Project file & How to reproduce

[stand.zip](https://github.com/user-attachments/files/18731724/stand.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0+win64

### Operating system

Windows 10

### Printer model

Prusa Mk4S with MMU3

u89djt, 2025-02-10:
This problem has been acknowledged and apparently will be fixed in the next release https://github.com/prusa3d/PrusaSlicer/issues/13814
The gcode output isn't affected, but it is an important bug because it is no possible to inspect the preview.

unkn-one, 2025-02-10:
As mentioned by @u89djt, duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13814

closed 2025-02-10 by unkn-one