prusa3d/PrusaSlicer #14114: MMU 3 Installation

Nebelhorntom, 2025-02-10:
Hallo,

habe einen Bausatz Prusa MK4S gekauft und erfolgreich zusammengebaut, Testdrucke funktionieren. Jetzt habe ich den MMU3-Bausatz zusammengebaut und montiert. Treiber und Updates für die Firmware habe ich heruntergeladen, die aktuelle Firmware für den MK4S ist installiert, Beim Versuch der Aktualisierung der Firmware für den MMU3 kommt immer wieder folgende Anzeige im Display: "Entlade Manuell - Unerwartetes Filament erkannt. Stellen Sie sicher, dass kein Filament geladen ist. Überpüfen Sie die Sensoren und die Verkabelung." Beim Drücken des Buttons "WDH." passiert nichts. Filament ist nicht geladen. Laut Bauanleitung auf S. 100 sollte die alte Filament-Sensor-Baugruppe entfernt werden. Nun befindet sich ja gar kein Filamentsensor mehr im Nextruder, oder?

Was ist hier das Problem? 

Für Hilfe wäre ich sehr dankbar.

Viele Grüße

Thomas Wolf

ojrik, 2025-02-11:
https://help.prusa3d.com/de/guide/9a-mk4-s-mk3-9-s-nextruder-mod_683106
https://help.prusa3d.com/de/guide/10a-mk4-s-mk3-9-s-einrichtung-und-kalibrierung_426826

Jan-Soustruznik, 2025-02-12:
Hello, @Nebelhorntom , based on your issue, I recommend reaching out to Prusa support, as the problem may be hardware-related. You can find instructions on how to contact our support team here: https://help.prusa3d.com/de/article/kundendienst_2287

closed 2025-02-12 by Jan-Soustruznik