prusa3d/PrusaSlicer #14127: Add option to disable scarf seems on overhangs

me25542, 2025-02-12:
**Is your feature request related to a problem? Please describe.**
Scarf seams, which have recently been implemented in PS, are great. They help hide seams, and they work well with default settings. The only area where they have not been reliable, in my limited experience, is overhangs. They (scarf seams) tend to make overhangs look much worse than regular seams.

**Describe the solution you'd like**
A way to use regular seems on overhangs (only when necessary; when there is not enough non-overhang perimeter to fit the scarf joint), and scarf seams on non-overhang perimeters. If *part* of a perimeter is an overhang, but enough of it is not an overhang to allow the scarf joint to fit, the seem should be placed there.

**Describe how it would work**
Add an option to "Disable scarf joints on overhangs". Possibly with the ability to set, in degrees, what is considered to be an overhang. If this overhang setting was not included, then the "Overhang threshold" setting from under "Support material" would be used.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
The ability to paint on scarf seams in the seam painting tool could also work (one could switch between painting normal seams and scarf seams, or enforce a certain type of seam in the painted location). One could, in some cases, use modifiers (height range or shapes) to shange is scarf seams are enabled during an overhang.

**Additional context**
I am using PS 2.9.0, on Windows 11 24h2, if it matters

neophyl, 2025-02-12:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13912

me25542, 2025-02-13:
Not quite a duplicate, as [#13912](https://github.com/prusa3d/PrusaSlicer/issues/13912) concerns bridges, not overhangs. But maybe it is close enough to not matter.