prusa3d/PrusaSlicer #14128: Pruscaslicer qui crash

marcp35, 2025-02-12:
Bonjour, 

voici mon problème des que je veux slicer peut importe ma pièce et sa taille sur Pruscaslicer celui crash,
Auriez vous des solution

merci a vous

neophyl, 2025-02-12:
How do you expect an answer when you haven't provided the most basic of information ?  What version of PS ?  What operating system ? No logs from your machine of the crash etc.

u89djt, 2025-02-13:
@marcp35 if you want someone at Prusa to look at your problem, they need the information specified in bug report template.  Those instructions are there to cater for persons with no experience of submitting bug reports such as yourself. Sadly, bug reporting takes significant effort because every user has a different situation, and very often, they only know that situation, and genuinely can't see how there could be more detail to give. For that reason, the bug report tells you what to do.
And users with some experience forget what it was like before they knew anything about what is happening.
The developers will ask you to write in English.

Si vous souhaitez que quelqu'un chez Prusa examine votre problème, il a besoin des informations spécifiées dans le modèle de rapport de bug.  Ces instructions sont là pour aider les personnes qui n'ont aucune expérience dans la soumission de rapports de bogues, comme vous. Malheureusement, le rapport de bugs nécessite des efforts considérables car chaque utilisateur est confronté à une situation différente et très souvent, il ne connaît que cette situation et ne voit vraiment pas comment il pourrait y avoir plus de détails à donner. Pour cette raison, le rapport de bug vous indique quoi faire.
Et les utilisateurs ayant une certaine expérience oublient à quoi cela ressemblait avant de savoir quoi que ce soit de ce qui se passe.
Les développeurs vous demanderont d'écrire en anglais. Utilisez Google Translate si vous en avez besoin.

marcp35, 2025-02-13:
Bonjour  
Merci déjà pour vos retour 
J'utilise Windows 11 
Voici ma configuration ordinateur  
Ryzen 5 3600x 
16G ram 
Carte graphique  rx6750xt 

Version PS 2.9.0 

Et malheureusement PS se coupe et aucun message d'erreur ne s'affiche 
Et t'il possible de trouver un rapport dans PS et si oui ou puige le trouver ? Afin de vous le transmettre 

Merci à vous

u89djt, 2025-02-13:
Step one: save the 3mf file just before you hit slice and it crashes. Zip the 3mf file and post it here. I'll see if it breaks for me.

kubispe1, 2025-02-18:
Hi, I am watching your discussion. The crash at the beginning of slicing makes me really sad, I would say, but based on the note, it is not possible to reproduce and then fix it.

Have you already tested a previous version? Does it make a difference?
Can you record the screen?

We will also release an alpha version soon. You can try it and let us know.

closed 2026-09-29 by prusaslicer-bot[bot]