prusa3d/PrusaSlicer #14129: Custom gcode insertion failing

amused578, 2025-02-12:
### Description of the bug

Version 2.9.0 on Windows 10.

Feature working fine in previous release.

Attempting to insert custom gcode of any type, let alone a simple PAUSE, fails.  Additionally, the system takes in the custom LCD prompt (M117) and does include it in the gcode output, but fails to display it in the sliced model view.

### Project file & How to reproduce

New Project.
Import STP/Step file.
Slice.
Move to sliced view.
Move slider to desired layer.
Right Click on '+' in red hexagon
Click on 'Add Pause'
Enter custom prompt in popup.
Click "OK"
Note that the gray hexagon does not change and that there remains a red hexagon with a '+' symbol at the layer point you just attempted to put in the Pause.
Reslice.
Note that the gray hexagon and red hexagon are unchanged.
Slide the red hexagon down to layer 1.
Note that the gray hexagon now shows a pair of vertical lines - the Pause symbol.
Hover over the Pause symbol.  The custom prompt is not shown, only empty quotes.
Export and examine gcode.
Note that M117 does exist with the custom prompt, however the PAUSE code (regardless of the variant, be it a G or M code, or macro) does not exist.
Attempt to print the file.
Note that the printer does not pause at any point in the print.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 10

### Printer model

Creality Ender 5 S1 running Klipper.

amused578, 2025-02-13:
You didn't read my ticket.  The function worked perfectly in 2.8.1. Adding gcode of any sort worked fine and printed without issue, doing exactly what the custom gcode (pause or otherwise) instructed the printer.  It does not work in 2.9.

I have both 2.8.1 and 2.9 installed on my machine.  I have compared these side by side (with only one version open at a time)  I can open a project in 2.8.1, slice, generate a gcode file and it works flawlessly.  I can open the same project in 2.9, slice, generate a gcode file and the custom inserted gcode does not happen.

When I attempt to attach a project file, I'm greeted with "unsupported file type: .3mf", ergo, I attempted to attach a project file, as requested.  This *system* did not permit the attaching of a project file.

[Here's a link to the project file on a google share.](https://drive.google.com/file/d/1Kh9-pIpVCFq9ltM8t4mvL-RpSk16fcWa/view?usp=sharing)

hitmusic100, 2025-03-06:
re "Attempting to insert custom gcode of any type fails" - agree

I can modify start G-code, but the modified start G-code is not included in the exported  G-code file.

amused578, 2025-03-06:
3 weeks and dev team hasn't commented nor has the ticket been assigned.  I'm now highly reconsidering my move to Prusa hardware.  The slicer is at least as important as the printer.  This is a critical feature for those of us who do *any* inserts into the print, require temperature changes for different filaments during a print, etc.  Most importantly, it was working fine in the previous production release and now fails to function.

neophyl, 2025-03-06:
Files can not be attached without zipping them.  That is why you couldn't attach a project file.

I opened up a new project for my old Mk3 printer, added a basic cylinder primitive, sliced and then inserted a custom comment at layer 6 using the insert custom gcode option.  I then sliced and exported it.

On opening up the gcode file in a text editor I search for my comment.

[image]

This was with 2.9.1 alpha1 on a windows 10 system.  So for me its all working.

I don't normally open files from google docs but in this case I did.  Your project for an nfc tag has a printer profile for a modified Ender 5 S1 correct ?  At least that's what I'm seeing.  

You have also inserted a Pause at Layer 11, is that also correct ?

When I go and actually look at your printer profile though, Printers>Custom Gcode, in the Pause field there is the following 

[image]

As you can see the Pause field is **empty**.  So what you are doing is inserting nothing.  This is why the Dev's ask for projects.  Its also why I suggested manually inserting a pause command using the klipper console to confirm the printer will actually PAUSE as klipper doesn't always come with a pause macro.  And THEN making sure that the matching Pause command is filled in to the Slicer field.  

Your printer profile has M600 for colour change but that is it.  It is no surprise that its not inserting anything is it.
If you had checked the items I suggested you could have easily figured this out for yourself.

amused578, 2025-03-06:
You sound like you're addressing some newb.  I've been programming professionally for 43 years.  You don't think I'd actually alter the printer profile to match Klipper and not Merlin, since Klipper has a built-in macro for PAUSE?   As for creating a .zip file, it'd be SUPREMELY helpful if something like that were noted.

Your attitude absolutely sucks.

If you're not associated with Prusa, they should bar you from this platform.  You're beyond abrasive, condescending, unhelpful, and based on a second comment confirming my issue, an utterly ignorant tool.

I'm closing this ticket.  Not because the issue is fixed, but because I refuse to deal with you further.

closed 2025-03-06 by amused578

neophyl, 2025-03-06:
Correct I am not associated with Prusa, just a long term user.  I do agree that the submission guidelines could do with updating to include information about zipping up files for attachment, but looks like Prusa are busy with trying to deal with thousands of open issues. 

While Klipper may come with a built in Pause Macro, a cursory google search shows that Creality's version which is not stock does not.  I found many reddit posts relating to this problem.  

I would have thought a 43 year professional would have checked all the usual resources before opening an issue.  And would have run some basic trouble shooting first.  I work with devs every day, half of them don't have the sense they were born with.  Just because you program doesn't mean anything.  Its not a stamp of quality.  You may be in the other group that can find their ass with both hands though, don't know as the internet is notoriously bad at conveying that sort of information.  25 year software test engineer btw.  So you can guess why my opinion of some Dev's is not the highest.

As for my attitude, don't care, its not a popularity contest.  You ask for help you take your chances on what's offered.  If you don't like the form it comes in then frankly tough.  Facts not feelings.

You should now have enough information to put the correct information into your printer profile to get it working as the problem area has been pointed out.

And the other comment 'confirming' your issue does nothing of the sort, looks like a completely different problem.  Until they supply the requested project file in their opened ticket though its going to be impossible to say for sure.  Again facts, no assumptions.