prusa3d/PrusaSlicer #14131: cant type into search box

jomo487, 2025-02-13:
Prusa Slicer 2.9.0
 Windows10 on Surface
the Search Box on the top right does not take text input.  Nothing displays while typing. Any words typed in take me to the Preference page.

Elump, 2025-02-14:
I have the same issue on my side.

Build: PrusaSlicer-2.9.0
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 11 (build 26100), 64-bit edition

kubispe1, 2025-02-20:
Hi, thanks for the report. We've already received similar reports and are aware of this issue. We are considering a fix along with other changes, but it won't be in the upcoming release—rather, sometime later.

wlgrd, 2025-08-02:
Any idea on when or what is causing the issue? I have two Windows 10 computers, and only one has this issue and I would love to edit any setting that may cause this.

hatchedegg, 2026-06-11:
I experience the same screen dependency. My laptop has the text field issue on its own 2K screen, but when I move PrusaSlicer to an external HD screen it works fine.

Thanks for the tip cobryan05! Hope it is resolved soon.

(Half year old laptop running W11 updated.)

SovereignGFC, 2026-09-02:
This is still a live issue and should remain under consideration for fixing.