prusa3d/PrusaSlicer #14132: An alternative implementation of top z contact for Support material.

MrBlackmidi, 2025-02-14:
**Is your feature request related to a problem? Please describe.**
https://youtu.be/2DmE34qT0Ls?si=Ip9qNyaT7-CBNkZz @34:46 Long one short, Matt Bahner describes it as a bug (which i don't think it really is, hence this being a feature request). Overhanging parts that had support material on it, the dimensions perpendicular to it are off by the set z contact (which is bad for large twisty puzzles)

**Describe the solution you'd like**
Rather than the support material being lower than the part by the set z contact distance, attempt to have the part that is being supported be offset by the set z contact distance without having the part raised by the set z contact distance.

**Describe how it would work**
Ditto

**Describe alternatives you've considered**
Matt Bahner mentioned about compensating for this via cad but its really tedious and slow hence this feature request which would make it a lot easier

ulab, 2025-04-11:
To me it looks like Matt did not remove the interface layer completely. 

[image]

If it were sagged down filament you wouldn't see the ends that clearly, because there would also be a perimeter around it.

MrBlackmidi, 2026-09-18:
h