prusa3d/PrusaSlicer #14133: wrong wipe way

spartik22, 2025-02-14:
wipe get out from material! See pict.
it's a problem for print quality
Pruslicer 2.9.0

[image]

u89djt, 2025-02-14:
You need to zip the 3mf project file to post it.

spartik22, 2025-02-14:
[Wrong way wipe.zip](https://github.com/user-attachments/files/18804509/Wrong.way.wipe.zip)

u89djt, 2025-02-14:
Confirming that there are wipes into mid-air.
@spartik22 what printer are you using, and what is your source of printer profile? I think the developers will need to know that if they're going to investigate. (I'm a fellow user.)

[image]

spartik22, 2025-02-14:
Ender 3 Max 
_what is your source of printer profile:_
I don't know exactly. Or I don't know where to find out. Probably Ender 3

u89djt, 2025-02-15:
OK, I think that's the extent of my triage efforts. Over to the wizards.

spartik22, 2025-02-15:
I'm not sure, but it didn't used to. Well, before 2.9.0

Jan-Soustruznik, 2025-03-12:
Hello, @spartik22,This issue no longer occurs in the latest release, 2.9.1, which was released yesterday. Please install the latest version of PrusaSlicer. I am closing this issue as resolved.

Have a nice day, and thank you!