prusa3d/PrusaSlicer #14135: Error manipulating SVG in Slicer 2.9.0

jmasterdude, 2025-02-15:
Slicer Continual Crash with Use Surface
Continual issue with the correct surface being used when it doesn't crash
Have simplified and reduced triangle count.
run Windows repair, despite no direction of a bad profile

Items from SVG are hanging in the air

PrusaSlicer
Version:   2.9.0
Build:     PrusaSlicer-2.9.0

[System Info.txt](https://github.com/user-attachments/files/18807124/System.Info.txt)

[Coffee cup sleeve POW.zip](https://github.com/user-attachments/files/18807128/Coffee.cup.sleeve.POW.zip)

u89djt, 2025-02-15:
Fellow user noting that you rembered to check "use surface" for the some svgs and and not others:

[image]

[image]

u89djt, 2025-02-15:
It looks like the individual elements of the svg have lost alignment or have been scaled a bit? You can see that the plaques are jumping up and down on the surrounding material That's obviously not what you want, and it will quickly get complicated, so might explain the crashes. Try tidying up the SVG and using it all at once?

[image]

jmasterdude, 2025-02-15:
The 'use surface' is what would cause the crash.

I would play with editing the svg, different positions, thicknesses etc, and change the order of the svg's as i mated the svg to the surface. I restarted countless times. This was the furthest I had gotten and didn't think I would be successful. I have since gotten further by rotating the model and getting the last items to use the face.

jmasterdude, 2025-02-15:
You also might be able to see that I'm trying to set multi materials. I had to break the svg into multiple files to be about to use the smart fill tool. I've had issues painting by hand, or having details simply fall off the print unless I used smart fill or merger the item into the main model. With use the surface with the svg, it wouldn't recognize the different areas, only the outline

u89djt, 2025-02-15:
If you change the type of all the svgs to "modifier" and set different tools, and use surface, you can see that they're not lined up. You'll need a small gap between regions in different svgs, I think, if you need to emboss? 

[image]

jmasterdude, 2025-02-15:
Modifier vs part? I did want the logo prominent, so figured I needed a part, not just modifier.

I can't look at this part for a bit, on different pc.

I've had to position each svg manually. Even though they are different exports from the same source svg, they seem to all have a different origin. I don't know enough about svg files to preserve the same origin for all of them. I exported them from inkscape.

image.png (view on web)<https://github.com/user-attachments/assets/796c3862-b86e-4d35-9bf2-949d35bb6451>

image.png (view on web)<https://github.com/user-attachments/assets/796c3862-b86e-4d35-9bf2-949d35bb6451>

u89djt, 2025-02-15:
I think this isn't a slicer issue. You're asking if to pile up mismatched svgs. There will be folk on the forums trying to do similar things. You can close the issue yourself.

closed 2026-09-29 by prusaslicer-bot[bot]