prusa3d/PrusaSlicer #14136: Perimeters are not supported by infill bridging

rcmaniac25, 2025-02-15:
### Description of the bug

A print where perimeters start inside the print, does not get supported and prints in mid-air.

There's existing logic where if a top layer needs to be printed
[image]
it makes some bridge infill to contact the existing infill
[image]
then does the solid infill on top of it
[image]

But, if it's a perimeter
[image]
it does none of that and prints in mid-air
[image]

In some cases
[image]
it outright prints in mid-air
[image]

The logic for this could also benefit two other issues I've seen with organic supports:

- Organic supports generated in mid air (I know there's a ticket for this somewhere, need to find it again). Basically: if it's known there's nothing under the printed part, add bridging somewhere so it can build on top of that.
- Perimeters that need supports get interface layers generated mid-air (causing a failure of both the supports and the print). I don't remember if this one exists already, but I have a project that I had to work around this issue.

### Project file & How to reproduce

This is from trying to print https://www.printables.com/model/1129607-metikoro-a-3d-printable-strategic-board-game-for-2

[metikoro.zip](https://github.com/user-attachments/files/18808751/metikoro.zip)

1. Open project
2. Slice it
3. On the first bed: Check layers 1.20-1.40 and 10.60-10.80 and you'll see mid-air printing of perimeters.

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Windows 10

### Printer model

Prusa MK4S

neophyl, 2025-02-15:
Looks like its because of the multi colour nature of the project.  If you change those features to match the extruder (2) of the underlying parts then it slices ok.

rcmaniac25, 2025-02-15:
Ah, that's a good point @neophyl . Though, given I want the multicolor, let me update title and such to indicate this bug applies to multicolor.

rcmaniac25, 2025-05-26:
Yea, still an issue in 2.9.2. Makes me worried on doing some detailed multicolor models if it's not going to support itself.

rcmaniac25, 2026-09-13:
Just tried this in 3.0-alpha11, issue is still there