prusa3d/PrusaSlicer #14138: prusa slicer software screen black

hayes-glitch, 2025-02-15:
when trying to click prusa connect to connect to my printer or after slicing to send to printer the screen loads with prusa filament icon for couple seconds then i just see the pop up window as a black screen. i restarted computer, closed and opened slicer, logged out logged back in, turned on and off printer, and re added it to prusa connect and nothing is working.

lukasmatena, 2025-02-18:
Can you please retest this in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1)? There is good chance that the problem will be gone, although there are multiple possible causes. Thanks.

hayes-glitch, 2025-02-18:
Which file type do I download to the flashdrive? I already tried the .dmg one and it didn't do anything.

thanks

hayes-glitch, 2025-02-18:
I wroked with chat support from prusa to remove the hardware completely then only download the zip folder and run it from github. it still has the black screen issue.

kocikdav, 2025-02-19:
Hello @hayes-glitch @sphyrnidae8 what OS are you using? Windows?

kocikdav, 2025-02-19:
Thanks, would you please try if reloading does do something? Edit -> Reload Web Content or ctrl+R.

closed 2026-09-29 by prusaslicer-bot[bot]