prusa3d/PrusaSlicer #14139: Bottom support interface generated when same layer has top support interface

Noisyfox, 2025-02-16:
### Description of the bug

1. Setting bottom interface layers to 0:
[image]
2. Placing things like this so there will be both bottom and top support interfaces at the same layer:
[image]
3. Slice, notice there is bottom support interface, even if it's set to 0:
[image]

### Project file & How to reproduce

[bottom-interface-prusa.zip](https://github.com/user-attachments/files/18813430/bottom-interface-prusa.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11

### Printer model

Prusa XL

Noisyfox, 2025-02-16:
I think this has something to do with this part of the code:
https://github.com/prusa3d/PrusaSlicer/blob/5c7888b11453eb96a95b49dfd4cbd3b56db64eb8/src/libslic3r/Support/SupportCommon.cpp#L1685-L1711

marcobeierer, 2026-09-17:
I just ran into this issue. It still exists in version 2.9.6, and I've also tested 3.0.0-alpha11 where I could reproduce it. Thus the legacy label can maybe be removed?