prusa3d/PrusaSlicer #14144: Slicer hangs on STEP import

smurfix, 2025-02-17:
### Description of the bug

```
$ prusa-slicer Hextraction\ Seek\ Tile\ step/Seek\ Tile\ Card.step
```
gets into an infinite loop, consumes 100% CPU forever.

The STL imported from the same Fusion3D source works without problems.

### Project file & How to reproduce

[hex.zip](https://github.com/user-attachments/files/18820348/hex.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Linux

### Printer model

any

u89djt, 2025-02-17:
(From a fellow user) Your step file just contains data describing a mesh.
[image]
The slicer opens it, it just takes a very long time because the mesh is encoded with a lot of extra information for every triangle to support efficient application of CAD editing processes.
[image]
Save the mesh as a mesh, and it imports as fast as you'd expect.
HOOPS exchange or whatever was using the libraries didn't convert the mesh into continuous surfaces. Remastering a mesh to the original surfaces currently takes user interpretation and labelling for an object with complex transitions between surfaces. The range of possibilities is huge, and you have to work out what the intent was.
There's no slicer issue here, so you can close this report yourself.

smurfix, 2025-02-17:
> The slicer opens it, it just takes a very long time because the mesh is encoded with a lot of extra information for every triangle to support efficient application of CAD editing processes.

Ah. Right. I seem to have been too impatient. On the other hand, 11 minutes at 100% CPU (1.8 GHz), does seem excessive for a <30-MB file.

In this case the importer might want to be a whole lot more efficient WRT skipping that information. It's a bog-standard Fusion STEP export after all.

At minimum I'd expect a progress dialog. Not ten minutes with no response whatsoever.

smurfix, 2025-02-17:
Report amended accordingly.

u89djt, 2025-02-17:
Your step file is not a Fusion export.
You can't just skip the information - it's how your triangles are being handled in the step file. They're not being handled as triangles per se.
The slicer has to reconstruct all the surfaces and connectivity for itself from all the cascades of geometric specification in the step format you gave it, and then begin to work out how to divide it up into triangles, because they are not stored in the step file explicitly as triangles.
This is the right thing to give to the slicer: 
[STLB ATF 12.4.0.zip](https://github.com/user-attachments/files/18827721/STLB.ATF.12.4.0.zip)
If you do export it from Fusion as a step file, it writes this:
[Fusion export Seek Tile Card.zip](https://github.com/user-attachments/files/18827955/Fusion.export.Seek.Tile.Card.zip)
That does export it as an explicit mesh using the COMPLEX_TRIANGULATED_FACE formalism.
Oddly, if I load that into the slicer, it decides that the object is of zero size. I would say that is a bug. If I add a brep box before exporting, the slicer shows the box but not the mesh. There's something to explore there. I'd need to see if it's reported elsewhere.

u89djt, 2025-02-17:
My description in my first comment was misleading - "extra" should have been something like "entirely different". Different quantity, different qualities, much of it of no value to a slicer.

closed 2026-09-29 by prusaslicer-bot[bot]