prusa3d/PrusaSlicer #14145: Option to remove both Printables and Physical Printers tabs

sgano, 2025-02-17:
### Description of the bug

Previous bug reports seem to be closed on this without regard to community inputs.    The Printables tab is basically a embedded browser and only adds bloat to the GUI for many users.  It also requires an internet connection which should be optional - thus there should be an option to hide/remove this tab.   Additionally the "Physical Printer" tab has never shown me anything other than "Something went wrong." with a reload button; however, I don't plan to login via the slicer so this tab seems completely unnecessary for some users and therefore the application should have the option to hide/remove it.  This would be a very simple option to add, therefore urge the Prusa Developers to reconsider this stance on insisting that these tabs always remain visible for everyone.  

### Project file & How to reproduce

open Prusa Slicer v2.9.0 - observe unneeded tabs making the GUI unnecessarily cluttered. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

All

### Printer model

NA

CRCinAU, 2025-02-18:
Just like the AppImage drama, they're just going to stick their head in the sand and ignore any of this.

Ignore that it drags in an entire web browser into the slicer.

Ignore the performance problems that non-exiting browser instances cause.

sorki, 2025-02-23:
Ideally the build process could detect missing `webkit` dependency and not include these parts.

I too would be happy with just an option to hide these buttons, similar to hiding log-in button which is way less problematic as it's on the other side of the window which makes it harder to click by accident. I've managed to hit `Physical printer` many times recently when trying to adjust printer properties and it causes a segfault of the whole app.

sithlord48, 2025-03-07:
Crazy idea save tons of built time and just use my web browser to open the web pages..

DiThi, 2025-03-20:
Please, this is needed, not just because we use them better with the web browser, also because it crashes the application (using latest Arch package), and I've lost several projects already because I clicked one of them by mistake.

MechanizedMedic, 2025-04-13:
Agreed, these should be optional features.

apopticon, 2025-07-12:
I surely don't need a rogue web browser in my slicer. Very disappointing for Prusa to have gone this route. In the meantime, rename or delete C:\Program Files\Prusa3D\PrusaSlicer\WebView2Loader.dll and those buttons will be rendered useless.

Fojtik, 2026-05-31:
This "feature" only adds ballast into PrusaSlicer. It makes application big. Slicer should be slicer and not www browser. Application that communicates on internet like browser, must be regullary actualised. It adds additional unwanted danger to PrusaSlicer.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena