prusa3d/PrusaSlicer #14148: STEP file shows a gap in model when sliced, but not in the same STL file.

mingabunga, 2025-02-18:
This is the same model, but when sliced in Prusa Slicer 2.9.0 the STEP file has a gap in the model - and crashes the printer. 

It looks like this. 

[image]

But when I slice the same file as an STL, then it comes out perfectly. 

[image]

Both files attached. 

[BIG DRUM MODIFIED v2.zip](https://github.com/user-attachments/files/18840124/BIG.DRUM.MODIFIED.v2.zip)

adamsbajgar, 2025-02-18:
The STEP file has open edges. Have PrusaSlicer repair them, and it will print fine.

TerPuc, 2025-02-19:
Hello @mingabunga, I have checked the STEP file and as @adamsbajgar mentioned, the file has open edges. Repair the model and it will work fine. There is no bug in slicer. I am closing this issue.

closed 2025-02-19 by TerPuc