prusa3d/PrusaSlicer #14152: Using PrusaSlicer 2.9.1's new sequential algorithum to plan sparse purge tower

rcmaniac25, 2025-02-18:
**Is your feature request related to a problem? Please describe.**
Sparse purge tower has been experimental for a while. It works well so long as you can ensure it won't collide with any print.

**Describe the solution you'd like**
PrusaSlicer 2.9.1 now adds a smart way to ensure it won't collide with other prints. Apply that to space purge towers so it can be placed/used without colliding with the print.

**Describe how it would work**
Turn on sparse purge towers and treat it as if it's a separate print object. If there's a collision, present an error. If not, then it's good.

**Describe alternatives you've considered**
Doing this by hand.

**Additional context**
I think the animation on the PrusaSlicer 2.9.1-alpha1 shows exactly how it would work, but one of those objects is a sparse purge tower.

lukasmatena, 2025-02-18:
This is actually in the planning. The problem is a bit more complicated than pretending that the wipe tower is an object, but we have certain ideas about how to get around it. However, the priority now is to make sure that the current implementation is bug free and can be built upon.

rcmaniac25, 2025-02-20:
Sounds good @lukasmatena . It just felt like an easy extra, though it does work different. And because it came to mind: Bonus points to do sequential multicolor prints with a sparse purge tower and not have to worry about collisions. I'll see if I remember to make a ticket for that.

rcmaniac25, 2026-09-13:
Still very relevant unless there's plans to drop single-nozzle multi-color support.