prusa3d/PrusaSlicer #14153: No Linux Build for PrusaSlicer 2.9.1-alpha1

P-C-R, 2025-02-18:
### Description of the bug

Can not find the Linux Build for PrusaSlicer 2.9.1-alpha1

### Project file & How to reproduce

Can not find the Linux Build for PrusaSlicer 2.9.1-alpha1

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.9.1-alpha1

### Operating system

Linux

### Printer model

Core One

Jan-Soustruznik, 2025-02-19:
Starting from version 2.9.0, we use Flathub for installing PrusaSlicer. Please install the Linux version via Flathub.

Here is a quick guide on how to use flatpak from the terminal. This is required to try out the alpha, beta and rc releases.
First, make sure you have flatpak installed and Flathub correctly set up. You can follow the steps at https://flatpak.org/setup/.

If you want to try out the alpha (such as this one), beta and rc versions you must add the flathub-beta repository (you only need to do this once):

```
flatpak remote-add --if-not-exists flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo
```
To install the latest version of PrusaSlicer:
```

flatpak install flathub com.prusa3d.PrusaSlicer      # for stable
flatpak install flathub-beta com.prusa3d.PrusaSlicer # for the unstable versions
```

To run PrusaSlicer from the terminal, you can do:

```
flatpak run com.prusa3d.PrusaSlicer       # for stable
flatpak run com.prusa3d.PrusaSlicer//beta # for the unstable versions
```

_Note that you can run all the mentioned commands with the --user option if you do not have the superuser privileges on your machine._

_Please understand that publication of PrusaSlicer on Flathub has to be preceded by the publication on our GitHub. This means that there will always be certain delay (typically couple of hours) before the new release shows up on Flathub. This is the expected behaviour, the extra time is needed to build the Flathub binary._

Ticked closed

closed 2025-02-19 by Jan-Soustruznik

P-C-R, 2025-02-19:
Thank you. Would be good to add this somewhere. I compiled the latest version yesterday myself but its not that convenient

salocain, 2025-02-19:
oooohhhh thank you , i looking for too.

kubispe1, 2025-02-20:
We will attach these few steps to a help page and include it in the release notes