prusa3d/PrusaSlicer #14154: PrusaSlicer Crash on Mac Mini M4 Pro.

rcludwick, 2025-02-19:
### Description of the bug

Prusaslicer crashes immediately upon launch.  Uninstalling/Reinstalling doesn't seem to fix the issue.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [43850]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.9.0 (PrusaSlicer-2.9.0)
Code Type:             ARM-64 (Native)
Parent Process:        zsh [739]
Responsible:           Terminal [577]
User ID:               501

Date/Time:             2025-02-18 19:23:06.7906 -0700
OS Version:            macOS 15.3 (24D60)
Report Version:        12
Anonymous UUID:        48442521-270D-702D-DF44-909CF1E21BBD

Sleep/Wake UUID:       9AFE0D49-B5F3-4CC9-B451-AED9D28E4C96

Time Awake Since Boot: 3800 seconds
Time Since Wake:       1506 seconds

System Integrity Protection: enabled

Crashed Thread:        17

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000b1d9430407e0 -> 0x000031d9430407e0 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x0000b1d9430407e0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [43850]

VM Region Info: 0x31d9430407e0 is not in any region.  Bytes after previous region: 54328165664737  Bytes before following region: 50743914264608
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  reserved VM address space (unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600020000000 [512.0M] rw-/rwx SM=PRV  

Thread 0:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18ce62f54 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x18ce75604 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x18ce6baf8 mach_msg_overwrite + 480
3   libsystem_kernel.dylib        	       0x18ce6329c mach_msg + 24
4   CoreFoundation                	       0x18cf8ca4c __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x18cf8b2ac __CFRunLoopRun + 1212
6   CoreFoundation                	       0x18cf8a734 CFRunLoopRunSpecific + 588
7   HIToolbox                     	       0x1984f9530 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x1984ff348 ReceiveNextEventCommon + 676
9   HIToolbox                     	       0x1984ff508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x190b02848 _DPSNextEvent + 660
11  AppKit                        	       0x191468c24 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
12  AppKit                        	       0x190af5874 -[NSApplication run] + 480
13  PrusaSlicer                   	       0x1053bcd30 wxGUIEventLoop::OSXDoRun() + 140
14  PrusaSlicer                   	       0x1054c0928 wxCFEventLoop::DoRun() + 40
15  PrusaSlicer                   	       0x1054252c0 wxEventLoopBase::Run() + 192
16  PrusaSlicer                   	       0x1053f5358 wxAppConsoleBase::MainLoop() + 208
17  PrusaSlicer                   	       0x10537caf4 wxApp::OnRun() + 36
18  PrusaSlicer                   	       0x10545bafc wxEntry(int&, wchar_t**) + 88
19  PrusaSlicer                   	       0x103691944 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 284
20  PrusaSlicer                   	       0x102ae8488 Slic3r::CLI::run(int, char**) + 22336
21  PrusaSlicer                   	       0x102aedfd0 main + 220
22  dyld                          	       0x18cb24274 start + 2840

[... 547 lines cut ...]
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.
```

### Project file & How to reproduce

No project available.  It crashes upon launch.

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9

### Operating system

MacOS 15.3

### Printer model

Prusa Core 1, Prusa Mk4s, Prusa Mk3S, Prusa Mini

lukasmatena, 2025-02-19:
Can you please zip and send us your config folder, if it does not contain anything confidential? The backtrace suggests that we might reproduce it if we have it.

Did you manage to run it at least once? Did it offer to migraye your 2.9.0 config into the PrusaSlicer-alpha directory, or not even that?

Thanks.

kocikdav, 2025-02-19:
Also if i see correctly, this happens on 2.9.0 release. Would you please try the newly released [2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1)?

rcludwick, 2025-02-19:
@lukasmatena 

I did this: 

```
$ find .//Library | grep -i prusa | while read file; do rm -rf $file; done 
```

So it should be a completely clean system.

@kocikdav 

I'll do that tonight.

rcludwick, 2025-02-19:
@kocikdav 

This seems to be fixed now.  Thanks.

lukasmatena, 2025-02-20:
Thanks. Closing.

closed 2025-02-20 by lukasmatena