prusa3d/PrusaSlicer #14155: MacOS: no bitmap handler for type 17 defined

rcludwick, 2025-02-19:
### Description of the bug

Upon starting a window comes up with something like this:

### Project file & How to reproduce

Start Prusaslicer from a clean install.  

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Sierra 15.3

### Printer model

Core 1, Mk4s, Mk3s, Mini

FidelCapo, 2025-02-19:
Could you please check whether this issue also occurs in version 2.9.1-alpha1? I believe this bug has already been fixed in that version. Thank you!

rcludwick, 2025-02-19:
@FidelCapo Yeah I'll check that tonight.

rcludwick, 2025-02-19:
Okay, so what seems to be happening was that I was running 2.8.0 and logged in with my Prusa account.  When I started 2.9.0 I got the crash detailed in #14154 and this bug.

So now with 2.9.1alpha I still get the message "No Bitmap Handler for type 17 defined".  If I don't click on the okay button within a few seconds, Prusaslicer will Lockup.  If I do click on it, Prusaslicer will appear to continue as normal.

Now I tried to do my best cleaning up all the data doing 

```
$ find .//Library | grep -i prusa | while read file; do rm -rf $file; done 
```

But apparently I was still logged in somehow when 2.9.0 and 2.9.1alpha started.  So I guess it wasn't as clean as I was hoping for.

The apparent fix was to log out of my account in prusaslicer, and that message goes away.  And things seem to work as normal.

rcludwick, 2025-02-19:
I was able to replicate this issue.  

1.  Wipe all configs in ~//Library   

2.  Start with 2.8.1.  And login using username/pw.  Then quit.

3.  Then install 2.9.1alpha1 and run that.  Import the 2.8.1 config.

4.  That message no bitmap handler for type 17 defined will pop up and it will completely become unresponsive after a few seconds.

5.  Requires a kill -9 to kill it.

kocikdav, 2025-02-24:
@rcludwick thank you so much for the repro notes. Can I ask you to try the alpha1 again now? There has been some changes on server side, which hopefully will change the format of file you are receiving. Potentially if there is a path to the bitmap, will it help to delete it?

Secondly, can i ask you to send us a zip file of your 2.8.1 config?

rcludwick, 2025-02-24:
Just tried 2.9.1-alpha1.  Comes up with that error and then locks up.

I'll get the zip file to you this evening if I can remember.

kocikdav, 2025-02-26:
Also when this message about no bitmap handler pops up, would you locate the file and send it to us?

rcludwick, 2025-03-01:
@kocikdav 

Hey, thanks for looking at this.  Apologies for not getting back to you sooner. 

This is the error text.
```
20:33:32: This is not a JPEG file.
20:33:32: Failed to load image from file "/var/folders/fx/7kdz7c5j0l7g7x0377h69hxm0000gn/T/prusaslicer-avatar-1840334539247780131.jpg".
20:33:32: no bitmap handler for type 17 defined.
20:33:32: This is not a JPEG file.
20:33:32: Failed to load image from file "/var/folders/fx/7kdz7c5j0l7g7x0377h69hxm0000gn/T/prusaslicer-avatar-1840334539247780131.jpg".
20:33:32: no bitmap handler for type 17 defined.
```

I copied all the avatars I have from that dir. 

[avatars.zip](https://github.com/user-attachments/files/19036642/avatars.zip)

rcludwick, 2025-03-01:
Some more info.  I deleted all the avatars.  Logged out.  Logged in.  Got the same error.  

The avatar is being downloaded from the server I guess, and then it's failing to decode it.  

```
rob@Robs-Mac-mini T % file *.jpg
prusaslicer-avatar-1840334539247780131.jpg: RIFF (little-endian) data, Web/P image, VP8 encoding, 40x40, Scaling: [none]x[none], YUV color, decoders should clamp
```

rcludwick, 2025-03-03:
2.9.1-beta1 doesn't seem to lock up.  So better I guess.

rcludwick, 2025-03-09:
My avatar is webp as well, btw.  It's been so long ago I don't remember if it was originally webp or not.

https://media.printables.com/media/auth/avatars/77/thumbs/cover/320x320/jpg/77291973-b915-4917-989c-8b3e11d3bfa1.webp

rcludwick, 2025-03-13:
Yeah.  I am seeing the same issue on 2.9.1.  

Logged out.  Deleted the Avatars.  Logged in.  Same error.

lukasmatena, 2025-03-14:
@rcludwick Thanks. We checked your url specifically and have a hunch that it may have been a caching issue, so we flushed the respective cache. Can you please retry now and let us know?

rcludwick, 2025-03-14:
@lukasmatena 

No luck.  It came back doing the same procedure I did a couple of days ago.

closed 2026-09-29 by prusaslicer-bot[bot]