prusa3d/PrusaSlicer #14158: Multimaterial painting for supports OR designating a part as support

alexsta001, 2025-02-19:
I work in aerospace and we are using a IDEX machine to print. Extruder 1 is using the material we want to print and extruder 2 is using soluble support material. It is important to know that the material cost **LESS THAN** the soluble support. For simplification purpose, let's say you want to print a U-shape tube like the one in the picture and you **HAVE TO** orient it the way it is. 

[image]

You would need to have supports inside the part and outside too. There are 3 ways to do the supports. 

1. Do them all in soluble material. While this will work, it will be very costly and will need a lot of tool changes.
[image]

2. Print only the interface in soluble support. While this will work perfectly for the outside support, it won't for the inside one. Because of the shape of the part, if I only do 3 interface layers of the soluble support, the non-soluble part of the support inside the printed part won't be able to come out. 
[image]

3. Exagerate and do something like 100 layers of interface. The problem with that is that not only it will be 100 layers of interface which is way more filled than regular support structure, but I will also have way to much interface layers for the outside support.
What would be perfect is to be able to **paint the support structure after it was sliced**. This way, I could keep the 3 layers of interface in soluble support for the outside support AND paint the support inside so that there is a middle part of the support done with extruder 1 that will be able to slide out once the soluble material is gone. _This would reduce the number of tool change, reduce the part cost and make the print faster._ 
[image]

So that would probably be the best of the best, but designating a part as a support would also be great. (Both would be just magic!!)
I was able to make something to accomplish what I'm describing. So I incorporated 2 other parts in the shape of the support structure that I wanted to be in the soluble material. I kept a 3 layers gap in the support I made and the part so that the slicer could make an interface out of the soluble support. This way, I was able to keep the 3 layer interface with soluble support in the settings and paint the support I made in CAD.

[image]

[image]

The result is this :

[image]

[image]

While this "works", it's not ideal. Because I wasn't able to select the supports I drew in CAD, it means that all the parameters of walls, infil, speeds, top, bottom, etc will be applied to the soluble support inside the part. If I would have been able to select the part and say "Print as support", it would have been a great alternative. 

SOOOOOO, to sum it all up, painting support structures would be the best solution. It would need the slicer to do a first slice to creat the supports and then redo a slice with the updated parameters of where to print with extrder 1 and extruder 2. As an alternative or a complement, designating parts as support would also work, just not as easily.

Thank you !

Jony01, 2025-05-27:
What about create help object inside tube (I draw 2d sketch, helper is red).

[image]

alexsta001, 2025-05-27:
@Jony01 Thanks for the response. There is a couple of issues:
- If I change the angle of the part on the printbed, I would need to redraw the support in CAD.
- Because of the shape of the support in red, it would be impossible to take the red part out even once the soluble interfaces are dissolved. One way it would work would be to extend the soluble part of the support on the top so that it goes down further (see picture)

[image]
For that scenario to work, we would still need to materialpaint the supports.

Also, the support you created makes me think of conical supports (like in Cura) that would really be the cherry on top. There is already another demand regarding conical support so I didn't included it here.

The idea behing material painting supports is that it would give the user complete control on when it's best to switch and when it's not. 

On last thing: I know that the supports are created more as 2D slices in the FDM module meaning that PrusaSlicer doesn't really compute a 3D mesh of the supports. However, the SLA module of PrusaSlicer creates supports with a 3D file and those can be exported with the part as a STL. In that regard, if the supports in FDM are created the same way that the SLA ones are, it would be possible to material paint the support (currently using a workaround by exporting those supports with the part and then re-importing them so you can materialpaint it). I don't know if it would work that way exactly, but the idea is there.

Let me know if oyu have other suggestions or questions !

Jony01, 2025-05-27:
> conical supports (like in Cura

Did you try [Organic supports](https://help.prusa3d.com/cs/article/organicke-podpory_480131)?

closed 2026-09-29 by prusaslicer-bot[bot]