prusa3d/PrusaSlicer #14160: Wiping into Air when perimiter is not a loop

VirtualGentlemen, 2025-02-19:
### Description of the bug

when extruding a circle, wiping while retracting works as expected. 
when extruding a line, wiping while retracting moves the extruder to beginning of extrusion line first, then runs the parimiter again. while moving, it wipes through air, making things worse!

### Project file & How to reproduce

can be reproduced using bency model.

[image]
[image]

[3DBenchy_wipe_example.zip](https://github.com/user-attachments/files/18873046/3DBenchy_wipe_example.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

windows 11

### Printer model

mk4s

VirtualGentlemen, 2025-02-19:
basically what is missing is the following exception: if detractions point is not identical with retraction, then the toolhead should wipe the extrusion path backwards.

VirtualGentlemen, 2025-02-19:
while we are at it, i cannot set a fixed length for wiping while extracting. please add, orca has it

VirtualGentlemen, 2025-02-19:
'wipe while retracting' is only enabled with nozzle sizes past 0.6mm. it CONTRIBUTES to strining. It may be the actual reason for the stringing fiasco of the PRUSA XL with the 0.6mm nozzles.

neophyl, 2025-02-19:
Might want to try your project in the 2.9.1 alpha that was released yesterday

[image]

VirtualGentlemen, 2025-02-19:
indeed, this issue is fixed in 2.9.1 and is a duplicate of #13753

closed 2025-02-20 by Jan-Soustruznik