prusa3d/PrusaSlicer #14164: last layer MMU3 print wipetower issue

dwprod38, 2025-02-20:
### Description of the bug

After finishing last layer of Dome part, printer continue to purge in air above wipe tower (see picture) instead of unload filament from extruder...

[image]

### Project file & How to reproduce

[FakeCamera.zip](https://github.com/user-attachments/files/18883847/FakeCamera.zip)

- Open BugWipeTowerEndPrint.3mf in zipped attached file
- Launch slicing
- See if you have same issue with "in air purge" above wipe tower

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Win11pro 24H2

### Printer model

Prusa MK4S

Jan-Soustruznik, 2025-02-20:
Hello, @dwprod38 , This is the intended and desired behavior. Closed

closed 2025-02-20 by Jan-Soustruznik

dwprod38, 2025-02-20:
Hello @Jan-Soustruznik 

> Hello, [@dwprod38](https://github.com/dwprod38) , This is the intended and desired behavior, see https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ) Issue closed.

This FAQ is obsolete since Oct 27, 2018 -->  no more solution have been solved until today (same issue with 2.1.1 and solved with 2.2.0 but bug is reappeared since 2.9.0) : see my comment/remark posted below...

https://github.com/prusa3d/PrusaSlicer/issues/1355#issuecomment-2671451741