prusa3d/PrusaSlicer #14167: request: set a fixed distance for 'wipe while retracting'

VirtualGentlemen, 2025-02-20:
**Is your feature request related to a problem? Please describe.**
Currently, the only way to have 'wipe while retracting' enabled, is to retract WHILE retracting. It is not possible to first retract, then wipe.

**Describe the solution you'd like**
I would love to have the possibility to set a fixed distance to wipe after retracting. see orca silcer implementation:

[image]

**Describe alternatives you've considered**
current wipe distance is a function of retraction distance, retraction speed and amout of retraction before wipe.
it is not possible to wipe after retraction.

Jan-Soustruznik, 2025-02-20:
Hello, @VirtualGentlemen  , thank you for your feedback! Could you provide some examples of use cases or clarify how the feature enhances printing? Have a nice day

VirtualGentlemen, 2025-02-20:
Hi @Jan-Soustruznik, 

i'm currently improving my retractions, and learning from printing a normal sized benchy with a 1.8mm nozzle in PETG.

When printing, i would like to minimize stringing and oozing on an extruded open ended line.

I would like to retract the material fast - 40mm/s
then, move the nozzle along the extruded line to a zone which is cooled down (2-3cm)
then lift.

this is currently not possible, only if set the retraction length to a long value and the retractions speed low.
however i would like to have the retraction length short (0.8mm), as i am printing with clear PETG and a HF CHT Nozzle. retracting further adds bubbles into the plastic when extruding again.
and i would like to retract fast to avoid blobs

i hope i explained my intentions clear enough, do ask me if i did not

closed 2026-09-29 by prusaslicer-bot[bot]