prusa3d/PrusaSlicer #14168: retraction dissapears if 'wipe while retracting' is set to 0%

VirtualGentlemen, 2025-02-20:
### Description of the bug

probably a UI bug, the purple retraction blob dissapears when 'wipe while retracting' was reduced from 1 % to 0 %

### Project file & How to reproduce

see images:

print settings are: 1 parimiter, no top and no bottom layers

[string test.zip](https://github.com/user-attachments/files/18891475/string.test.zip)

[image]
[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1 alpha1

### Operating system

win11

### Printer model

mk4s

Jan-Soustruznik, 2025-02-21:
Hello, @VirtualGentlemen, if you set "Retract amount before wipe" to 0%, no retraction is performed before the wipe. So, I believe this is the correct behavior.

closed 2025-02-21 by Jan-Soustruznik

VirtualGentlemen, 2025-02-21:
Hello, @Jan-Soustruznik if i disable "Retract amount before wipe" then the retractions appear. if i enable and set to 0% (same as disable), they dissapear.