prusa3d/PrusaSlicer #14169: extrusion overlap changes on single parimiter objects depending if one or two parimiters are set to print

VirtualGentlemen, 2025-02-20:
### Description of the bug

depending on if i set to print with 1 or 2 parameters, (no infill and no top and bottom layers), the retraction and detraction overlap or not. see attatched pictures. 'wipe while retracting' must be turned off.

[image]

[image]

is this a parameter in prusa slicer? even if not, i would expect it not to change if the object is already a single paramter object. 

### Project file & How to reproduce

[string test.zip](https://github.com/user-attachments/files/18891553/string.test.zip)

switch between 1 and 2 paramter printing, and see retraction position change

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1 alpha1

### Operating system

win11

### Printer model

mk4s

VirtualGentlemen, 2025-02-20:
this overlap does not happen in orca slicer, even with 2 parimiter printing.
 so i assume this is a bug.

[image]

Jan-Soustruznik, 2025-02-21:
Hello, @VirtualGentlemen, thank you for your feedback. This is an optimization of the seams we imported in [PrusaSlicer 2.8.0-alpha5](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-alpha5).(for more info you can see the changelog) This improvement will help minimize seam visibility and reduce non-print moves. Have a nice day