prusa3d/PrusaSlicer #14170: Don't use the word "register"

mproctor13, 2025-02-20:
### Description of the bug

The verbiage:

"Do you wish to register download from Printables.com" can be misinterpreted. Saying "Do you wish to link downloads from Printables.com"

and "Downloads can be linked to only 1 executable at a time". Would be more acceptable to many privacy minded people. 

[image]

### Project file & How to reproduce

Just start PrusaSlicer for the first time.

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Does not matter.

### Printer model

Does not matter.

kubispe1, 2025-02-21:
I have to say, you might be right. However, as I see it, we are already using 'registered' in this context in several other places, including our help section. So, for now, we will not change it.

I hope we can improve the overall wording, notifications, and descriptions with some bigger UI changes in the future. Thanks for your opinion!

closed 2025-02-21 by kubispe1

daneasley, 2026-02-10:
Please do give this further consideration.

Top three search results are users like me wondering what on earth we're signing up (registering) for.  Then there's this: https://help.prusa3d.com/article/opening-models-in-prusaslicer-from-supported-websites_399198 - which doesn't use the word register at all.  

Why not add a sentence or to to the dialog itself explaining what it is the user is agreeing to?

I don't think I'm a naive user - I've spent the last eight years managing a makerspace full of 3d printers.  Had absolutely no idea what this dialog meant.