prusa3d/PrusaSlicer #14172: "Communication with Prusa Account is taking longer than expected" while logged in

murk-sy, 2025-02-21:
### Description of the bug

This message appears after I log into Prusa Account (and seems to continue). Pressing stop logs me out even though I can send prints to Connect and even use the Connect tab.

[image]

Status icons work and so does sending prints
[image]

### Project file & How to reproduce

 Don't know how to reproduce but it's happening for me all along with 2.9.1-alpha1

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1-alpha1

### Operating system

Windows 10

### Printer model

Mini+, XL

Tholi78, 2025-02-21:
same here (mac os)

murk-sy, 2025-02-22:
Extra info:
in the first instance the semaphore is visible:

[image]

While in the second one it is not (while Prusa Connect still works and account name is shown)

[image]

kocikdav, 2025-02-26:
Hello @murk-sy, thank you for reporting. This message is supposed to inform you about some failing communication with servers, in your case probably getting information about available printers. But there shouldn't be a case where you have this notification while the services are working. We need to look at it.

SPE-2704

evaneaston, 2025-03-10:
As is confirmed in the relate issue's comments, @cristian-rivera 's work around of setting a profile photo worked for me.

ormandj, 2025-03-11:
I also was able to solve the problem via the Avatar change. One point of note, before this fix, when existing PrusaSlicer on a MBP M4, the application would actually hang and had to be force quit. Once the avatar was there, it was able to exit normally, so it sounds like the application was blocking on whatever call it's making that requests that avatar.

lukasmatena, 2025-03-12:
Thanks. This is a duplicate of #14219 (which should now be fixed by PrusaAccount update). Closing this one (#14219 has more info in it).