prusa3d/PrusaSlicer #14173: vase mode layer hight

kurabolna, 2025-02-21:
Hello, it is about printing in vase mode and consistency of layer hight
For instance: you are printing in vase mode with 
layer hight 3mm and
first layer hight 2.5mm

you also enabled 3 bottom solid layers

I would argue that in this instance you have two separate proceses first 3 layers are normal "layer by layer print" and on 4th layer, the vase mode (spiral printing) starts.

What hapens is that on 4th layer the spiral print starts at zero hight (and not at 3mm) relative to the 4th layer (the top most layer of the bottom solid layers) and than gradualy spirals up.
That causes layer below to be pressed down and causes an "elephant foot" until it reaches 3mm.
It should keep layer hight consistent at 3mm thought both proceses.

I would think that the same thing happens in reverse if you have solid top layers selected, but am unable to check.

I only became aware of this isue because i print in clay with nozzles of 2mm and up to 6mm and sice clay is way softer than plastic those things become wery apparent. Vase mode is the most common mode of printing with clay printers and this for me is quite an issue.

This is also the topic on Prusa forum
https://forum.prusa3d.com/forum/general-software-discussion/spiral-printing-clay/#post-211320
https://forum.prusa3d.com/forum/prusaslicer/print-profiles-for-wasp-clay-printer-elephant-foot-problems/#post-737362

Thank you and best regards

u89djt, 2025-02-21:
You need to zip a 3mf file to post it.

kurabolna, 2025-02-24:
Hello, i added prusa slicer files, simplifz 3d gcode file and some photos froim test prints from both sliced files.

Thank zou and best regards.

u89djt, 2025-02-24:
This is from me, a fellow user, so don't be surprised if a dev swoops in and corrects me:
You need to activate "Use relative E distances" in the printer tab, and add "G92 E0" to the layer custom gcode
[image]
[image]

kurabolna, 2025-02-24:
Thank you fellow user, will try and report back.

kurabolna, 2025-02-24:
Dear sir Fellow User, you are a genius...... It werks! You have made me wery happy. I will report this soluttion to the relevant posts on prusa forum.
Thank you again.

u89djt, 2025-02-24:
Wonderful! The solution came from trying every setting I could find until something changed. It came up in an issue reporting the step on the top layer a while ago. https://github.com/prusa3d/PrusaSlicer/issues/13355

u89djt, 2025-02-25:
The G92 E0 setting is what the slicer told me to do in an error box when I set the Relative E thing. It wouldn't let me slice until I'd entered it. That suggests that the slicer could potentially fill-in that change, but bear in mind that it doesn't know what else you might try to do in the layer gcode, so it's your responsibility, I guess.
Another issue over the last couple of days shows that the gcode is case sensitive, so it has to be capital G and capital E.

TerPuc, 2025-03-03:
Dear @kurabolna, I close this issue as I understand it has been solved.

closed 2025-03-03 by TerPuc