prusa3d/PrusaSlicer #14182: 2.9.0 Highly inefficient print paths

NielschB, 2025-02-21:
### Description of the bug

In some scenes the slicer produces highly inefficient print paths.
With attached example it seems that the algorithm is "scanning" print areas in vertical direction resulting in a lot of unnecessary travel moves. (Note that it picks the first spot at the very top, then very bottom, then back to top and then working its way down resulting in long X travel moves)

[image]

### Project file & How to reproduce

[3.3mf.txt](https://github.com/user-attachments/files/18914937/3.3mf.txt)

Just slice the project and pick a layer to see the travelings

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 Win64

### Operating system

Windows 11 24H2

### Printer model

Custom build (Based on mk3 / MMU3)

murk-sy, 2025-02-24:
Should be resolved in the 2.9.1 release

Screenshot from 2.9.1-alpha1:
[image]

NielschB, 2025-02-24:
I have checked 2.9.1_a1 version and can confirm that this is resolved.
Thank you

closed 2025-02-24 by NielschB