prusa3d/PrusaSlicer #14190: Scarf joint placement set to Contrours or Everywhere generates relative Z-values in gcode

davidkonz, 2025-02-23:
When "scarf joint placement" is set to a value other than "Nowhere", the Z-value after a "WIPE_END" will be a relative value and results in print failure on the Ender3 v2.

Attached are the project files and gcodes generated with only this setting enabled and disabled.

Search for "Z-.009" in both gcode files and notice it only occurs in the "scarf.gcode" file.

[prusa_scarf.zip](https://github.com/user-attachments/files/18927197/prusa_scarf.zip)

Jan-Soustruznik, 2025-02-24:
Hello, @davidkonz , thank you for your report. I created internal issue(SPE-2701). We will try to fix in some of next release.

lukasmatena, 2025-02-27:
Fixed in [2.9.1-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-beta1). Closing.

closed 2025-02-27 by lukasmatena